mirsadm/motioncam #110: Samsung S20 (snapdragon) does not reach 24 FPS

ELKILI, 2022-01-03:
Samsung S20 (snapdragon) does not reach 24 FPS in any case.  It doesn't matter how many output FPS you choose, 24/25 / or 30. You will never get past 22 FPS.  Several users experienced the same problem, trying all possible configurations.

2blackbar, 2022-01-03:
Did you tried to actually sync the sound  with video and conclude its 22fps precisely? What app says in export menu is not a real info about actual fps.
Some of my clips show its 5fps but its actually perfect 24 after conversion

ELKILI, 2022-01-03:
Yes, I have noticed by the audio and also by the number of FPS resulting, that multiplying frames by the time recording, missing. I have checked with a timer also. Moreover what I can fix down speed clip resolve but what more worries me is the lack of fluency video and small jumps giving.

taranish, 2022-01-05:
Same issue on S10 (Exynos), can't get past 21-22fps, no matter if I use RAW12 or RAW16, compressed or uncompressed. Even with maximum crop and pixel binning there is no way to get above 22fps.

mirsadm, 2022-01-05:
One test that would be helpful is that if you go into "Burst" mode and take a picture, what are the timestamps in the pictures?
For eg:
0.0s, 0.03s, 0.06s etc

taranish, 2022-01-05:
> One test that would be helpful is that if you go into "Burst" mode and take a picture, what are the timestamps in the pictures? For eg: 0.0s, 0.03s, 0.06s etc

On my S10 it's 0,00, -0,03, -0,07, -0,13, -0,17, -0,23, -0,27, -0,30, -0,37, -0,40, -0,47, -0,53, -0,57, -0,60, -0,63, -0,70

mirsadm, 2022-01-05:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta5/MotionCam.8.0.6-beta5.apk

can you give that one a go please?

taranish, 2022-01-05:
> https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta5/MotionCam.8.0.6-beta5.apk
> 
> can you give that one a go please?

No change on my S10, raw video is still 21-22 fps and burst created 21 images in a second.

mirsadm, 2022-01-05:
> On my S10 it's 0,00, -0,03, -0,07, -0,13, -0,17, -0,23, -0,27, -0,30, -0,37, -0,40, -0,47, -0,53, -0,57, -0,60, -0,63, -0,70

Interestingly here it is producing 30 fps but dropping some frames. Can you try this one to see if anything changes:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta5/MotionCam-release.apk

Deadcode85, 2022-01-06:
Hi, S20 Exynos user here.
8.0.0 beta 5 burst:
0.00, -0,07, -0,10, -0,13,-0,17 , -0,23, -0,30, -0,33, -0,37, -0,40, -0,47, -0,50, -0,57, -0,60, -0,63, -0,70, -0,73, -0,77, -0,80, -0,83, -0,90, -0,97

128GB version, factory defaults, only non retail app is Motion Cam.

RAW video: with beta 5 now the fps counter is stick to 22 fps instead of bouncing between 19-21. On the live view i can see frame drops all the time not just when recording video.
Max crop and max crop + pixel bining is 22 fps too.

Deadcode85, 2022-01-06:
After few tests, it's most likely 18 OR 20 fps, it's "decided" at start. Doesent matter if the desired fps is set to 24 or 30p

ELKILI, 2022-01-06:
Samsung S20 ultra (snapdragon)
0,00-0,03-0,07-0,10-0,17-0,23
V 8.0.6 beta 5

taranish, 2022-01-06:
> Interestingly here it is producing 30 fps but dropping some frames. Can you try this one to see if anything changes: [...]

Tried this one as well on the S10, got exactly 22 frames in burst mode between 0.00sec and 1.00sec.
RAW video records 22fps.