mirsadm/motioncam #112: The converter does not work with AMD FX 8300 CPU

Gogo1973, 2022-01-04:
The converter does not work with AMD FX processor, with RYZEN no problem, Windows 10 21H2. Am I not doing something right, or do I need something extra?

mirsadm, 2022-01-04:
It is probably because they are missing some needed CPU instructions. Next time I update the tools I'll recompile it so that it works on older CPUs.

Gogo1973, 2022-01-04:
Thanks,for everything you have done so far with MotionCam!

mirsadm, 2022-01-04:
Can you see if this fixes it:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta4/motioncam-tools-0.0.4-windows.zip

Gogo1973, 2022-01-04:
No, same as 0.0.3 doesn't work ...

Gogo1973, 2022-01-04:
Strangely, the second file started, and both are made with version 8.0.6-beta4

mirsadm, 2022-01-04:
ah I uploaded it to the same beta which is probably why it seems confusing. great, i'll close this

closed 2022-01-04 by mirsadm