mirsadm/motioncam #117: MotionCam Tools

Gooner447, 2022-01-06:
Having issues with MotionCam tools 0.0.3. dmg  App closes down when I click the start convert tab, using macOS Monterey
 Ver 12.1

mirsadm, 2022-01-06:
does it happen every time? if so, if you could upload one of the container files somewhere tht would be helpful.
Also, are you using an older Mac by any chance?

Gooner447, 2022-01-06:
Yeah it happens everytime on my mac but I also have a window laptop and I tried the same container and it worked fine.  Your next question, "then why not use the laptop". answer: it"s very slow and takes ages whereas my mac mini is blisteringly fast 😀

mirsadm, 2022-01-06:
haha no that's not my next question at all, I'd like to get the bottom of it since I use a Mac. Is it a new Mac Mini (M1) ?

Gooner447, 2022-01-06:
Yeah, with 16gb  internal memory. my new toy, fast as f**k.   I was try out an 8k clip from my samsung on Davinci and it handles it no problem, without proxies or optimised medea.  Only slow thing about it is me, trying to keep up. lol

mirsadm, 2022-01-06:
Give this one a go:
https://github.com/mirsadm/motioncam/releases/download/8.0.6-beta6/motioncam-tools-mac-0.0.5.dmg

They're built for Intel but should still work on the new Macs

Gooner447, 2022-01-07:
That one didn't work, crashes the app. I tried container file on windows laptop and it worked fine.  Mac won't open the app unless I  Ctrl/click on the file in the applications tab in finder and make an exception

sbrindignigni, 2022-01-07:
the tool crashes before loading the preview with different containers (Windows 10, tool v0.0.3). I share one

https://drive.google.com/file/d/1kNXKvuWf9P4w5RhxSoUq7LepVsVtHw4r/view?usp=sharing

Gooner447, 2022-01-08:
Good morning, I have tried a few more times but the app keeps crashing, I'm wondering if its an issue with the M1 chip but I can use the win 10 laptop, until it's fixed .  However, is there any chance you could fix the dropped frame issue with Samsungs by lunchtime and maybe cure covid for good.  From reading the demands and request from people on here and Telegram, this should be no big deal for you to get done lol

mirsadm, 2022-01-11:
@Gooner447 haha on it. I need to buy an M1 Mac myself I think

mirsadm, 2022-01-15:
@Gooner447 I've built the tools for the M1 here:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta4/motioncam-tools-mac-m1-0.1.1.dmg

Gooner447, 2022-01-16:
Hi, just tried getting this error message
[image]

mirsadm, 2022-01-16:
hmm ok I will repackage it tonight

mirsadm, 2022-01-16:
What MacOS version is your Intel & M1 running? I wonder if its something to do with that. I've compiled them again with lower minimum versions here:
https://github.com/mirsadm/motioncam/releases/edit/8.0.7-beta5

Gooner447, 2022-01-16:
[image]

Can you tell me which phone, that's availabe in the UK, that will work completly and have the use of all the apps functions

mirsadm, 2022-01-16:
Does it still start but crashes when you click convert to dng? I wonder if its something to do with permissions. If you go into Security & Privacy under Files and Folders. Is motioncam-tools in there? I've rebuilt the tools again in the post above, the run on the same mac + os version for me so bit baffled by whats going on.

As for a phone recommendation that's a tricky question since I've only got access to two phones :) I use a OnePlus 7 Pro which works with everything but it's getting old now. I believe the Pixel 4 also works.

Gooner447, 2022-01-16:
Its not showing in privacy files and folders.  when I try to open it it just comes up with this error, some of the earlier versiond opened up but crashed when I hit the convert to dmg
[image]

Gooner447, 2022-01-16:
As for the phone, thanks , I'il try on telegram and see if anyone there has any more info

LuckyBastard73, 2022-01-19:
It would really be great to get the preview in the correct aspect ratio.
It is hard for me to determine if I see a clip fom the main lens or the ultra wide angle if everthing is transformed to 4:3.
A floatable (detachable) and resizable window could be a huge improvement as well.

mirsadm, 2022-01-19:
@LuckyBastard73 yes these are all good ideas unfortunately it's just me working on this and I still have a full time job to do during the day :)

the specification for the container file is within the code at the moment. to be honest I'm not happy with the format, I created it very quickly and it has some shortcomings. My intention is to replace it with something more robust in the near future.

@Gooner447 I have bought a new M1 MBP and can reproduce the problem so should be able to sort this out soon

LuckyBastard73, 2022-01-20:
Fulltime jobs are thight;-)
Creating the perfect container format: super easy, barely an inconvenience;-)

Based on: https://itsryangeorge.com/portfolio/pitch-meetings/

Gooner447, 2022-01-20:
Ok Thanks, have fun with MBP

Gooner447, 2022-01-20:
I don't know if you do much film editing and colour grading but since I got the M1, I have been editing the DMG files in Lightroom Classic it's super fast and there is lens correction, de-noising plus all the usual controls. Then it's just a case of syncing the files and rendering them as DMG's , into Davinci and bob's (you know the rest)

Gooner447, 2022-01-30:
Hi, have you had any joy with MC Tools for the M1 chip. There was a message in the user guide, I didnt notice before.

_The app has been modified, and its code does not match the original signed code._

mirsadm, 2022-01-30:
Yep I'll release them again soon. I'll sign them with a developer ID this time so it should hopefully work

Gooner447, 2022-01-30:
Great thanks

mirsadm, 2022-02-07:
Excellent I'll close this.

closed 2022-02-07 by mirsadm