mirsadm/motioncam #118: Xiaomi Mi11 viewfinder freezes after 3 seconds

Jeneeeeek, 2022-01-06:
The problem is on the xiaomi device, mi11, sdm865 , codename Venus.
The problem is that the device requires a different viewfinder or something like that, in Google camera it uses 3+ level 2 api, For the viewfinder to work, the "OPmode" session is turned on, it is called eis v3(61448 or 0xF008 in HEX)

https://user-images.githubusercontent.com/96690234/148452059-bec78411-e1f0-484b-b51a-ad14ca067bf3.mp4

mirsadm, 2022-01-07:
I've been looking into this. Need to figure out how this translates to the Camera2 API

Jeneeeeek, 2022-01-07:
> Я изучал это. Необходимо выяснить, как это переводится в Camera2 API.

invoke-virtual/range {v0 .. v5}, Landroid/hardware/camera2/CameraDevice;->createCustomCaptureSession(Landroid/hardware/camera2/params/InputConfiguration;Ljava/util/List;ILandroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)

    invoke-virtual {v0, p1, v1, p3}, Landroid/hardware/camera2/CameraDevice;->createCaptureSessionByOutputConfigurations(Ljava/util/List;Landroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)V

maybe it will give you a hint

mirsadm, 2022-01-09:
I have found where this is set but unfortunately it's not accessible from the NDK. The app starts the camera from C++ and it seems like the session operation mode is hardcoded to "NORMAL" there. I've contacted Samsung, I'm not holding my breath there. I'll keep digging into this.

Jeneeeeek, 2022-01-09:
> I have found where this is set but unfortunately it's not accessible from the NDK. The app starts the camera from C++ and it seems like the session operation mode is hardcoded to "NORMAL" there. I've contacted Samsung, I'm not holding my breath there. I'll keep digging into this.

perhaps this is a problem with the buffer, and the manufacturer must solve it himself, since most devices on 888 do not have such a problem