mirsadm/motioncam #119: Fps becomes stable and lock on after few seconds

amitkattal, 2022-01-07:
If i chose 24 fps and press record, it will begin by recording 25 fps and after like 6 seconds becomes 24 fps and then can continue easily. Sometimes it even begins with 30fps and then changing to 24 fps after 6 seconds. I believe maybe this is also why the audio doesnt sync in post.

2blackbar, 2022-01-07:
Yeah i noticed that on all framerates it goes to desired fps +1 and then stabilises to desired fps after a few seconds.
so on 24 it records 25 for a few secs , then goes to 24, with 15 it records 16 then goes to 15 etc...

mirsadm, 2022-01-07:
What you're seeing is the rate at which the frames are being written to storage not the final frame rate. I realise this is confusing a lot of people so I will change that. Basically if the memory slider is not full, you shoud be getting a consistent frame rate output

closed 2022-01-11 by mirsadm