mirsadm/motioncam #123: Option (or default behavior) to remember last used mode (video vs. photo)

darkblaze69, 2022-01-07:
It would be more natural and of great convenience if the app remembers previously used mode (video vs. photo).

Right now by default it is in Photo mode (or Night photo mode) when opening the app. And to shoot only videos during video sessions one must always select "Raw video" every time the app is opened. One could miss great moments when shooting videos while forgetting about active default Photo mode.

nerdpaolino, 2022-01-07:
I need to reset also fps, iso and shutter every time I open the app. Is it possible to remember last used settings? Thanks

2blackbar, 2022-01-07:
would be handy i agree , also access to settongs when im video mode, now its only in photo.

mirsadm, 2022-01-07:
@nerdpaolino reset in what way?

and yes it should remember the mode

darkblaze69, 2022-01-07:
At least in video mode after closing app it resets OIS preference (I switch it off) and also "unsupported" frame rate is not stored between launches.

mirsadm, 2022-01-07:
The unsupported one is intentional because some people have mentioned the app crashes if they select certain options. Then the app will crash on start up. It use to be saved but I removed it for that reason. 

OIS should be saved though

darkblaze69, 2022-01-07:
ok, thanks, make sense about unsupported then.
Switching off OIS doesn't persist between app launches.

nerdpaolino, 2022-01-08:
Every time I open the app the settings are:  fps 30 iso 2000 aperture 1/100. I need to manually change them to my preferences

2blackbar, 2022-01-08:
Yeah would be much faster if app remembered last used shutter speed, iso, and if WB,AE and focus were locked  or not so we dont have to change them over and over when opening app.Its currently most time consuming.
Kinda like actual manual mode that doesnt switch back to Auto exposure once its set to manual (iso or shutter is locked).
I get that some stuff can lock the app in crash loop , but for fixing that we have clean cache function in android app menus.

mirsadm, 2022-01-08:
Updated so last capture mode is saved here:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta0/MotionCam.8.0.7-beta0.apk

I could save the manual settings but I'm not sure that makes sense? Maybe save them for an hour or two then reset them. After a little bit of time it wouldn't make sense to apply the same settings since the exposure has likely changed significantly.

2blackbar, 2022-01-09:
Well, in professional cameras when shooting RAW video , people stick to lowest noisy iso like 50 in case of phones, shutter 48 and 24 fps, we dont really change shutter much, mostly when fps changes, then use nd filters and manually adjust iso if scene change requires, thats what most raw filmmakers do.
Its totally fine for raw video mode section to not change shutter and iso at all unless we do it manually or switch to AE.
Exposure Is most of the time locked to manual and we dont really use auto exposure.Cause AE forces noisier isos with lower dynamic ranges.
New version gives me more stable recordings , doesnt switch up fps , it doesnt remember manual shutter and iso when i enter conversion menu and go back to raw video record, it would save time if it would, we go to converter sectiom to delete videos and record again but then its AE again and its constant screen tapping, with locked values it would save time, i find its most time consuming task to set shutter and iso to 48/50 eveeytime im back from converter menu.

darkblaze69, 2022-01-09:
Thanks!, video mode now persists.
but OIS toggle still not saved.

mirsadm, 2022-01-09:
Yep I'll save all the settings next update

nerdpaolino, 2022-01-09:
Thank you very much!

mirsadm, 2022-01-09:
Iso/shutter/ois is saved now
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta1/MotionCam.8.0.7-beta1.apk

closed 2022-01-09 by mirsadm