mirsadm/motioncam #125: when developing ravs in the mobile lightroom, the grid is visible

ScherbakovAl, 2022-01-09:
I am using mi 8 light.  When developing a dng file in a mobile lightroom, a mesh appears on areas with a smooth gradient.  It is visible well enough.  And another question .. why can't you take a photo with full sensor resolution?

2blackbar, 2022-01-09:
IS this happening with other apps for raw photo ? like opencamera, firstlight,moment,hedgecamera ? Try them out, maybe its a deeper issue.

ScherbakovAl, 2022-01-09:
Unfortunately no other application can capture raw from the camera of this phone model.  Therefore, I cannot compare this with other cameras.  If you increase sharpness and detail to the maximum values ​​in a mobile lightroom, then the lattice structure is very clearly visible.  It's strange that the resolution of the picture taken in motioncam is lower than the resolution of the matrix in the phone.
[image]

2blackbar, 2022-01-10:
Post dngs

ScherbakovAl, 2022-01-10:
https://www.dropbox.com/s/da1izxjxtz0i9ln/1641837815660.dng?dl=0
Apparently not the main camera is used.  Therefore, the resolution of the picture is strange, there is no way to change focus, there is no way to switch cameras (including the front one)