mirsadm/motioncam #126: Saved images are too small and soft/unsharpened

nerdpaolino, 2022-01-09:
Recording videos at 3840x2160 I get frames not bigger than 4.4Mb, sometime less. I thought that this it was due to the applied compression, but now I'm not so sure anymore.

Yesterday @lordsvarogofmax posted this image: [https://drive.google.com/file/d/1M8lHTe5x1ciPaffw05fau2VV0sCa2oJ1/view?usp=sharingive.google.com/file/d/1M8lHTe5x1ciPaffw05fau2VV0sCa2oJ1/view?usp=sharing](Gdrive). It is much more sharper and is 8Mb.

Today I've opened another file: [https://storage.rcs-rds.ro/content/links/c8f8a96e-9b09-469d-8eb0-1fe99dc80a86/files/get/frame-005982.dng?path=%2Fframe-005982.dng](DNG)
Same size and same sharpness.

Probably with an image so soft (initially I've thought it was out of focus) there are less data to save. I've noticed the same problem with the photo posted today by @ScherbakovAl for the bayer grid problem. He has a Xiaomi phone too. I suspect that this is related to Xiaomi brand.

mirsadm, 2022-01-09:
Are you able to capture a DNG with the stock camera as a comparison?

nerdpaolino, 2022-01-09:
Stock camera deliver a much sharper raw, much bigger in size, same behavior in opencamera.

In motioncam I tried saving DNG along jpeg in photo mode and I get 2 different results. Raw is blurry, Jpeg is sharper as it should be.

nerdpaolino, 2022-01-09:
Hera are two sample images:

[https://drive.google.com/file/d/1Xd6LW1ojLb5oWitxrFy-c7AzQi-aVog_/view?usp=sharing](shot1)

[https://drive.google.com/file/d/1wQBBqbnKhufED8ANfjpVSDxf7RfDo6Pe/view?usp=sharing](shot2)

mirsadm, 2022-01-09:
@nerdpaolino the photo mode applies sharpening. There's something not quite right about the DNGs you posted though. Some of these issues are almost impossible to fix though. There's nothing particularly special about the way MotionCam uses the camera so more often than not it boils down to the implementation of the manufacturer being broken

nerdpaolino, 2022-01-09:
Yes, I'll try to overcome this issue applying sharpening in post. Thank you very much for your time.

mirsadm, 2022-01-09:
Selecting RAW16 in the settings may change the behaviour so it's worth a shot

nerdpaolino, 2022-01-09:
I will try this first time I'll have a chanche to go outside.

ScherbakovAl, 2022-01-09:
And I like the softness of the raw.  It lacks the oversharpness that stock cameras add.  I believe that this only spoils the image.  On mi 8 light, I can't turn on raw16.  It drops to raw12.

fabiorug, 2022-01-09:
With root of the phone it can change. Or use camera2api opencamera, camera fv5

2blackbar, 2022-01-09:
Yeah it drops to raw12 but once I choose raw16 then I notice that it actually records bigger files than raw12 so it might be resetting itself just when you reenter options again