mirsadm/motioncam #128: motioncam tools crashes when preview large containers - output folder is empty

icysummer99, 2022-01-10:
motioncam tools crashes when selecting large container in tools window- unable to see preview details
Only ocurring with containers larger than about 5GB
Also, if I don't attempt to see preview and try and convert - the conversion fails with an empty folder

Note: I've tried in both MacOS and Windows 10 - same issue is occurring in both OS. 
The containers convert with no issue on the motioncam Android app.

I've included some of the error output from the Mac version- let me know if you need the full output.
Process:               motioncam-tools [1732]
Path:                  /Applications/motioncam-tools.app/Contents/MacOS/motioncam-tools
Identifier:            com.motion.motioncam-tools
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           motioncam-tools [1732]
User ID:               501

Date/Time:             2022-01-10 22:55:47.834 +1100
OS Version:            macOS 11.6.2 (20G314)
Report Version:        12
Anonymous UUID:        1799A291-6ABB-894F-FE67-BA3EE626D5F0

Time Awake Since Boot: 500 seconds

System Integrity Protection: enabled

Crashed Thread:        32  Thread (pooled)

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type motioncam::IOException: Invalid end chunk

icysummer99, 2022-01-10:
[motioncam_tools.txt](https://github.com/mirsadm/motioncam/files/7838721/motioncam_tools.txt)
Full error output attached

mirsadm, 2022-01-10:
Thanks for reporting that, will take a look. Did you have the same issue converting it on the phone as well?

icysummer99, 2022-01-11:
> Thanks for reporting that, will take a look.

I did more testing. Seems the containers got corrupted/truncated during transfer from phone to pc.
I was transferring from phone internal storage > phone sd card > pc hard drive.
I will find a more reliable phone app for transferring the containers.

mirsadm, 2022-01-11:
OK will close this for now.

closed 2022-01-11 by mirsadm