mirsadm/motioncam #132: NOT WORKING MBA M1 12.1

MotleyTRIP, 2022-01-11:
NOT WORKING MBA M1 12.1

mirsadm, 2022-01-13:
what exactly is not working?

MotleyTRIP, 2022-01-14:
0.1 version still crash when after import try to select for further action!!!

mirsadm, 2022-01-15:
I've built them for the M1 Mac here:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta4/motioncam-tools-mac-m1-0.1.1.dmg

MotleyTRIP, 2022-01-16:
m1 versión doesn't even start ))))))))))) intel starts but still with above issue

Mrdoumhann, 2022-01-18:
Thanks for the m1 support. But I couldn't get it to work either. I tested it on M1 Air (Big Sur) and I get the the "application is damaged" error.

mirsadm, 2022-01-18:
Yep I will fix this soon. Ordering an M1 Mac for testing :)

mirsadm, 2022-01-19:
Finally got a new M1 MBP. I can reproduce the problem so will fix soon.

mirsadm, 2022-03-17:
@Mrdoumhann @MotleyTRIP this is now finally fixed in the new betas

closed 2022-03-17 by mirsadm