mirsadm/motioncam #136: [Feature Request] Bluetooth Release Support

LuckyBastard73, 2022-01-11:
I have a Bluetooth shutter release and that works with virtually every camera app right out of the box.
MotionCam seems to be an exception. Or let's say it does not work as expected. 
I always get a burst and the editing tools no matter in which mode I am.
I would like to get an image if I am in night or normal photo mode.
A burst only if I am in the burst mode.

mirsadm, 2022-01-11:
Sounds like it works by simulating the volume up/down keys. That'll explain why that happens.

LuckyBastard73, 2022-01-11:
You are right but why is this not the case for GCam, the stock camera app oder OpenCamera?
I would expect the same behaviour.
On the release there is an large and a small button. 
They behave different, as one takes the burst and the other lets the volume control appear.

mirsadm, 2022-01-11:
It happens because I wasn't aware that's what most people expected out of the buttons :D I can easily change it to replicate that behaviour

experimentator1709, 2022-01-11:
@mirsadm 
https://github.com/mirsadm/motioncam/issues/49

mirsadm, 2022-01-13:
^ This will be fixed in next update.

LuckyBastard73, 2022-01-15:
I can confirm that the remote shutter release is now working the same way as in most other camera apps in the latest beta.

closed 2022-01-15 by LuckyBastard73