mirsadm/motioncam #147: [Feature Request] DNG conferter for Linux

nerdpaolino, 2022-01-15:
Is it possible to have the DNG converter also for Linux machine?
Many thanks.

mirsadm, 2022-01-15:
Yes it is possible to compile it to run on Linux and I do intend on doing that. I'm already stretched for time supporting three platforms though. The good news is someone has reported that they work well under Wine so if you don't want to wait give that a go.

nerdpaolino, 2022-01-15:
I need to wait because I wasn't be able to run it under wine, thanks.

darkblaze69, 2022-01-15:
> 

Runs perfectly under wine. Just install vcrun2017, e.g winetricks vcrun2017.

nerdpaolino, 2022-01-15:
I'm installing it now. Thanks for your help.

nerdpaolino, 2022-01-15:
I've installed with winetricks vcrun2017, but when i launch wine motioncam-tools.exe it complains that it doesn't find a lot of dll's.