mirsadm/motioncam #148: Asus ROG Phone II

6GHOST6FACE6, 2022-01-15:
In any video modes i have only 10 fps. I try crop to 4k, but it is only 10fps. I try all fps in video recording, but i have only 10fps. And in no crop mode program shows me resolution 8000x6000. I see that my cpu is not overloaded, or even loaded for 50%.

mirsadm, 2022-02-04:
I think I have finally solved this one. Check out the latest beta, it should pick a reasonable resolution

6GHOST6FACE6, 2022-02-07:
I i tested, in this version i have 3 dots for sensors, but i have only 2, wide angle camera and main. In wide angle i have 30 fps on all settings higher than 30 fps, crop and no crop. In main camera i have unbelievable resolution 8000 x 6000, that is wrong i have only 48mp camera. And on main sensor on all settings i have 10fps. And i see in my monitoring, that i have only 4% GPU load, and 95% load CPU.
[image]