mirsadm/motioncam #149: 8.0.7 beta 4 crashes on Google Pixel

nocantsin, 2022-01-16:
8.0.7 beta 4 crash on a Google Pixel (6 Pro) with the same symptoms and error message as in the past: After starting the app, the camera display remains black except for the UI elements, and after a few seconds, the app crashes with the pop-up message "Error: The camera encounter an error".

A log file is attached here.

[motion_cam-8.0.7-beta4-crash_log.txt.gz](https://github.com/mirsadm/motioncam/files/7876203/motion_cam-8.0.7-beta4-crash_log.txt.gz)

mirsadm, 2022-01-16:
Thanks for letting me know. Did the previous beta work? That will narrow down the changes

2blackbar, 2022-01-16:
pixel 2xl works fine if that helps

mirsadm, 2022-01-16:
Did the previous beta work? That's narrow it down

nocantsin, 2022-01-16:
The previous beta worked!

mirsadm, 2022-01-16:
Beta 3 is ok? Beta 4 no good? That narrows it down for me I'll sort it out today

nocantsin, 2022-01-16:
Sorry, was mistaken (tested again). Beta 3 has the same issue, Beta 2 works.

mirsadm, 2022-01-16:
Ah great know exactly what it is. The Pixel phones seem extremely finicky

mirsadm, 2022-01-16:
@nocantsin should be fixed now:
https://github.com/mirsadm/motioncam/releases/download/8.0.7-beta5/MotionCam-8.0.7-beta5.apk

nocantsin, 2022-01-16:
Yes, this version works! So many trhanks for all your great work!

mirsadm, 2022-01-16:
Excellent :)

closed 2022-01-16 by mirsadm