mirsadm/motioncam #152: Information loss indicator in dark areas

lordsvarogofmax, 2022-01-18:
Good afternoon!
A very good tool is implemented in the application - we can track the translucent zones using the clipping indicator.
Dear developer, please make an indicator of information loss in dark areas so that we can avoid too dark areas if possible.
If too white places are marked in red, let too dark be blue. For example)

mirsadm, 2022-01-18:
@lordsvarogofmax here you go:
https://github.com/mirsadm/motioncam/releases/download/8.0.8-beta0/MotionCam.8.0.8-beta0.apk

I've switched the colours for both so they are friendly to colour blind people

amitkattal, 2022-01-19:
Could there be an option to switch the dark indicator on or off? because sometimes when i press record, the colors stay stuck on the preview window.

mirsadm, 2022-01-19:
yep I'll fix that one in the next update, it shouldn't get stuck like that