mirsadm/motioncam #153: Crash of Convert/Export Windows tool

LuckyBastard73, 2022-01-18:
When I try to open a 19GB clip the converter/exporter on my Windows 10 (21H2, x64) machine (16BG RAM) crashes.
In my case this happened with version 0.1.1.

mirsadm, 2022-01-18:
Does this happen on all files or just that particular one?

LuckyBastard73, 2022-01-18:
Only with that file. Next next bigger one I have is only 12GB.
When I star the app in WinDbg I get the following info:
[image]
The crash happens when I press here
[image]

LuckyBastard73, 2022-01-19:
To be honest I don't want to stress my phone with the export of such large files.

LuckyBastard73, 2022-01-19:
I tried the export on a second machine where I copied the file to directly from the phone and here it works.
Maybe the container file got corrupted during the copy process.
So the crash does not seem to happen with regular clips.
An app should not crash in such a situation anyway but as this is not a commercial product this is completely fine.

closed 2022-01-19 by LuckyBastard73