mirsadm/motioncam #154: The quality of the merging in the brust mode is worse than in the merging of Raw video

Jeneeeeek, 2022-01-18:
hello, I did a test, shot a 1 second video, and then set the 
merging of 16 frames, the result came out better than when merge 16 shots in burst mode.
suggestion to move merge from Raw video to photo(burst) mode too 

[image]
[image]

LuckyBastard73, 2022-01-19:
I am pretty happy with the merging on the ZTE Axon 30 Ultra.
If it would really change an option to keep the old algorithm would be great.
The suggested number of frames for merging are in my opinion a little too conservative.

Considering that theese images are from the smaller ultra wide angle sensor
[image]
[image]
[image]

mirsadm, 2022-01-19:
It uses the same algorithm but in photo mode it is a more conservative. I wanted to be sure that it doesn't introduce any artifacts. In the video mode those issues become less important because the artifacts are less visible due to motion.