mirsadm/motioncam #157: (!!!) Broken conversion in the application 8.0.8-beta0 or Windows tool [RAW video]

experimentator1709, 2022-01-19:
On a smartphone, stacking is 4 frames. On Windows, the preview closes, and after the convert - the folder is empty.

https://drive.google.com/drive/folders/1O01BiX8qmLLYNJRYL3pWyK_DjjsH6g_A?usp=sharing

mirsadm, 2022-01-19:
This seems to be ok on my machine. Are you saying it crashes after clicking create preview?

lordsvarogofmax, 2022-01-19:
Greetings! When converting on a smartphone and activating noise reduction for 4 frames and above, I encountered this effect. At the same time, when converting without noise reduction and converting via a computer, DNGs are obtained without damage

lordsvarogofmax, 2022-01-19:
Most likely the problem is in updating the apk
https://drive.google.com/drive/folders/1rTkosW3VVxngJHk-zYYsmCXY21xMhM7j?usp=sharing

mirsadm, 2022-01-19:
I'll look into this, do you mean after updating the APK it is not longer an issue?

lordsvarogofmax, 2022-01-19:
No. I mean, this problem appeared in the update 8.0.8-beta0.

mirsadm, 2022-01-19:
Oh I see, will take a look

experimentator1709, 2022-02-02:
@mirsadm 
Strange, but I downloaded my video_20220119_185646.0.container and it opened on the preview in all versions of tool...

Are you fixed noise reduction by stacking in app?