mirsadm/motioncam #166: MotionCam Tools error on W7: ... entry point CreateDXGIFactory2 could not be located in ... dxgi.dll

newillusion, 2022-01-21:
There is short solution on QT site: [QT solution](https://forum.qt.io/topic/78380/entry-point-not-found-the-procedure-entry-point-createdxgifactory2-could-not-be-located-in-the-dynamic-link-library-dxgi-dll)

...

>  Add this:
> 
> `-no-feature-d3d12`
> 
> to your configure call.
> 
> I used in full:
> 
> `configure.bat -prefix "C:\Qt\5.9\msvc2015_static" -release -platform win32-msvc -confirm-license -opensource -nomake examples -nomake tests -nomake tools -static -static-runtime -opengl dynamic -angle -combined-angle-lib -no-feature-d3d12
> `

2blackbar, 2022-01-22:
So it makes converter work on win7 ? Thats great, when i was on win7 i couldnt run converter.

mirsadm, 2022-01-22:
Thanks I think next time I update the tools I'll also compile using and older version of QT which should support windows 7

newillusion, 2022-01-22:
Huge thanks!

mirsadm, 2022-01-28:
I've updated the tools to use QT 5 which should work with Windows 7:
https://github.com/mirsadm/motioncam/releases/download/8.0.8-beta1/motioncam-tools-win-0.1.2.zip

newillusion, 2022-02-07:
Thanks! Now the tool works great!

closed 2022-02-07 by newillusion