mirsadm/motioncam #167: container name fps

kot0991, 2022-01-22:
Suggestion to make a feature that would add a k/s mode to the name of the container?
For example at the end of the container name 230-24fps or 230-60fps
Right now I'm looking at the folders I shot a week ago and I can't remember what frame rate I shot at. Would be a very useful feature.

*Thanks*

*DeepL used by translate*

mirsadm, 2022-01-22:
Yes that is a good idea. I might save a little text file along the video with that information.

2blackbar, 2022-01-22:
Yeah i guess it is a different workflow when we cant playback the clips in realtime easily to see if its slowmo one or regular speed.
I dont mind extra info in names, nobody keeps the original names anyway when editing but it would be useful.