mirsadm/motioncam #170: Black viewfinder in newer versions LG G7

hshsa, 2022-01-27:
Hello, in the latest version that I download on December 30th it worked perfectly fine but I needed the feature that keeps the last opened mode 

So I downloaded all betas from beta 0 8.0.7 and in all versions the viewfinder is black. Captured photos are normal however

mirsadm, 2022-01-27:
Do you use dual exposure? In photo mode you can see the viewfinder but not in video mode?

hshsa, 2022-01-27:
No, I don't use dual exposure. In 8.0.4 viewfinder is NOT black but if dual exposure is on than the viewfinder is black

In 8.0.7 with default settings the viewfinder is black at all mods (night, burst, video, photo). I tried to switch on dual exposure in this version hoping that it will fix the black viewfinder but it didn't.

mirsadm, 2022-01-27:
Is this still a problem? If so I'll reopen the issue

hshsa, 2022-01-27:
Yes, it's an issue, reopen please

I want to use your new versions because of the "last opened mode" feature but I can't because of black viewfinder

mirsadm, 2022-01-27:
does switching cameras help?

hshsa, 2022-01-30:
I switched to front and then back - still black viewfinder even on the front camera. There's no buttons to switch to ultra wide camera if you ask about that

mirsadm, 2022-02-04:
I am hoping this is fixed in the latest beta. Let me know how you go.

hshsa, 2022-03-26:
> I am hoping this is fixed in the latest beta. Let me know how you go.

Hello, nope. I also noticed there's no more DE (dual exposure) toggle in settings. In 8.0.4 (the version with normal viewfinder) it caused black screen if turned on, in 8.0.7 viewfinder was black by default although the DE was off and in the newest version the viewfinder is still black + DE is permanent, can't even turn it off/on

The app takes photographs though, no problem with that