mirsadm/motioncam #172: Compressed raw crashing windows tools 11

2blackbar, 2022-01-28:
I recorded compressed , it crashed when trying to preview it, cant export it, but when recorded uncompressed raw, it works fine .
Would be great to get GPU /CUDA acceleration in the future to speed up process on pc.

mirsadm, 2022-01-28:
I've updated both the tools and the app. This was a problem the tools using some instructions that may not be available. You can give it a go here:
https://github.com/mirsadm/motioncam/releases/download/8.0.8-beta1/motioncam-tools-win-0.1.2.zip

You'll need to use the new app as well:
https://github.com/mirsadm/motioncam/releases/download/8.0.8-beta1/MotionCam.8.0.8-beta1.apk

Compression should work now

2blackbar, 2022-01-29:
It shows proper preview with crop and aspect ratio , converts as well but it stays on "starting up" dialogue and doesnt show if its actually converting (it does i checked files in folder) so its kinda stuck on that small window until it finishes conversion.
So id say its fixed, just that small issue persists, sometimes id like to see how fast its converting the files, its not a big deal tho.
Dialogue "starting up" might be understood as program being stuck when its not.

2blackbar, 2022-01-29:
It works on win7 on my laptop but its crashing when trying to preview frames, it converts fine tho when clicking convert button without checking frames.

mirsadm, 2022-02-05:
Cheers, I'll need to put up a note for that somewhere

closed 2022-02-05 by mirsadm