mirsadm/motioncam #174: 8.0.8-beta1 vignette correction

FilipZamec, 2022-01-28:
Thanks for the update.

Vignette correction fixes the vignette, but leaves the color shift intact on Mi 11 ultra, so I still get pinkish center.

mirsadm, 2022-01-29:
would you mind sharing a very short container file? that'll help me fix it

FilipZamec, 2022-01-29:
Sure. Here you go.
[https://drive.google.com/file/d/1UAs2qVVWkkD-6MO13vIwkzTPcFgsJhM1/view?usp=sharing](url)

mirsadm, 2022-01-29:
Could you make an uncropped recording and also take a DNG from your stock camera app please? I can't really reproduce this with my phone.

FilipZamec, 2022-01-30:
Hi, thanks for taking your time. You can't reproduce it, because most phones don't have color shift issue :( 
Anyway this is StockCamera DNG and Uncropped from MotionCam.
Thank you

[Stock DNG](https://drive.google.com/file/d/1UX6cnMYAQaFXKkZgoT52AlEK6df9fQ1S/view?usp=sharing)
[MotionCam uncropped](https://drive.google.com/file/d/1UvbJA08amEK_9vrahy_AbWaijgidVW_M/view?usp=sharing)

mirsadm, 2022-01-30:
Is it something that happens with other apps as well? If it's just MotionCam that'll be good since it's something I can fix

FilipZamec, 2022-01-30:
Ok, made some tests and here it is.
Stock cam takes the same dngs as Open camera or Freedcam, They have size always over 20 MB. 
MotionCam has sizes from 4 MB to 11 MB depending on the scene. I don't know if it matters.

But if I open Open Camera DNG in Photoshop, it removes vignette and color shift. 
If I do the same with Motion Cam DNG, it removes the vignette, but not the color shift. 

Also this is what it looks like after vignette correction in Open Camera:
[image]

and this is what it looks like in MotionCam:
[image]
There is this pinkish center.

All files can be found here:
[files](https://drive.google.com/drive/folders/15_QHZkQwsKZZLu6vXE5DSGA2B979fQ-C?usp=sharing)

Thank you...

mirsadm, 2022-01-30:
Ok figured this one out. The bayer arrangement on your phone is GBRG and the two green channels were in the wrong order. Will put out a fix soon.

FilipZamec, 2022-01-30:
You're a star 🌟

mirsadm, 2022-02-04:
Finally updated. Check out the new tools/app, should fix this at last.

FilipZamec, 2022-02-05:
Thank you Mirsad.
Color shift is finally gone. Even from uncorrected vignette 👍
But now it seems that Highlight Recovery in Davinci does absolutely nothing and it seems like I lost Dynamic Range. Will test it further...

experimentator1709, 2022-02-05:
@FilipZamec 
exposure up :)

FilipZamec, 2022-02-05:
After corrected exposure I still have blown out highlights which can't be recovered. In previous versions it was recoverable. 
I will check it later what's changed...

mirsadm, 2022-02-05:
I think I can fix this one. Looks like the correct range of pixels is not restored so it doesn't think it's clipped.

FilipZamec, 2022-02-05:
You are right :)
This is old version.
NotRecovered
[image]

And recovered
[image]

Now it looks always like not recovered...

Thank you, appreciate your work. Tell us finally how we can donate to your wonderful project :)

mirsadm, 2022-03-17:
@FilipZamec this has finally been fixed in the new beta

closed 2022-03-17 by mirsadm