mirsadm/motioncam #176: Too much lens

gzewani, 2022-01-31:
Smartphones don't have that many lenses. There is no connection to some new lenses or the picture slows down.
In addition, the telephoto lens completely stopped working. On previous versions it works.
**8.0.8-beta1**

[image]

MarlogGit, 2022-02-01:
Same problem on Xiaomi Mi9Tpro (Snapdragon 855) Lineage os. Previous version OK.

mirsadm, 2022-02-01:
Yep I'll fix that next update.

nousername1, 2022-02-01:
Hi
I have an Huawei P30 Pro its possible to use the Main Sensor since the have the best Quality, Night Vision.
Currently I am not sure what I should use the "normal" Camera Software who use the Main Sensor or your Software who use Raw but a low quality Wide Angle Lense...
By the way do you want some Video I took?
Sorry when I post here so much but maybe its possible to add an Information when the Software detect an Mono or an Stereo Microfon. It seem the recognise the Stereo Boya Adapter as Stereo but I am not sure right now.
Thanks

gzewani, 2022-02-01:
> Yep I'll fix that next update.

When will it be possible to allocate more RAM for recording? As I understand it, this will allow you to shoot video longer without losing frames. For example, I have 12GB of RAM. At the moment I can shoot 6016 * 4512 only for about 9 seconds, then frames are lost.

nousername1, 2022-02-01:
Is there a way to chance to amount of Ram the Software can use?

FilipZamec, 2022-02-01:
> > Yep I'll fix that next update.
> 
> When will it be possible to allocate more RAM for recording? As I understand it, this will allow you to shoot video longer without losing frames. For example, I have 12GB of RAM. At the moment I can shoot 6016 * 4512 only for about 9 seconds, then frames are lost.

Be aware. I have the best results using only 1024 ram. I can shoot 60 fps without frame drops. If I use more ram, U have more framedrops.
Test it 🙂

nousername1, 2022-02-01:
Where can I set the Ram level?

gzewani, 2022-02-01:
> Be aware. I have the best results using only 1024 ram. I can shoot 60 fps without frame drops. If I use more ram, U have more framedrops.

Don't work :(

> Where can I set the Ram level?

In settings

> In photo mode there is settings button.
> But the resolution you're shooting in is high. And storage write speed isn't fast enough.

How do you know in what resolution he records video? He just went to my topic and asked where I change the amount of RAM. A little higher, I answered him that the amount of RAM is configured in the settings.

2blackbar, 2022-02-02:
ram wont help much with speed, use ssd drive and split recordimg option, run app called cpdt , it will tell you recording speed of your phone, leave compression on 2

gzewani, 2022-02-02:
> ram wont help much with speed, use ssd drive and split recordimg option, run app called cpdt , it will tell you recording speed of your phone, leave compression on 2

What's the advice for noobs?
Firstly, not everyone will be able to use the SSD due to USB 2.0
Secondly, the cpdt does not always see the external drive. There are other analogues.
Third, why do you think level 2 compression is optimal? Why deliberately enlarge the recorded video?

nousername1, 2022-02-02:
My "Problem" is the USB Port on the Phone is blocked by my Mic Adapter.

mirsadm, 2022-02-04:
The lens issue should be fixed in the latest update