mirsadm/motioncam #177: Thank You!

HACKERALERT, 2022-02-02:
I just wanted to say thank you for this amazing app. Before I knew of MotionCam, I believed that Apple's ProRes is many tiers ahead of what Androids can do and that there is no competition with Apple's superiority. But with MotionCam, we finally have a capable competitor to ProRes, which can capture truly RAW footage with no limits other than the phone's storage speed. I've done some testing on my midrange Samsung and the quality is simply unrealistic under good lighting. The amount of potential with your app is unprecedented and I'm really excited to see what's coming! Keep up the good work, MotionCam has the potential to bump Android's camera landscape to a whole new level.

PS: Where can I donate a few bucks to support this project? I and probably many others would love to support your work ✌️.

experimentator1709, 2022-02-02:
I agree. Support project

mirsadm, 2022-02-02:
Thanks very much I appreciate the feedback! I haven't set up any donations for the time being, I am still thinking about that. In the mean time if you could rate it on the Google Play store that would be very helpful!
https://play.google.com/store/apps/details?id=com.motioncam&hl=en_GB&gl=US

experimentator1709, 2022-02-03:
@mirsadm 
People leave feedback, but they are not visible ... maybe premitioning is included?

newillusion, 2022-02-07:
That is great work! My phone's video quality gets second life. Huge thanks for author.