mirsadm/motioncam #180: 8.0.7 on Samsung S20 FE 4G (SD) worked great under A11 ... with A12 the app crashes a second after start

giteos, 2022-02-03:
I already deleted the Cache / AppData, rebooted the device, re-installed from Google Play Store (8.0.7) but the problem persists.
(I am not able to check non Play Store versions hence the device is locked due to company security restrictions -> no app installation from other sources.)

Gogo1973, 2022-02-03:
I have the same phone with the A12 and there is no problem, I even always install the application from here!

giteos, 2022-02-03:
> I have the same phone with the A12 and there is no problem, I even always install the application from here!

My S20 FE is a SM-G780G ... which app version so you use?

giteos, 2022-02-03:
Got it working again after some hustle. The following did the trick for me:

1. Shut down into Safe Mode (hold shut down button for longer then the safe mode is offered, Motion Cam app can not be used in this mode)
2. Setting -> Apps -> Motion Cam -> Force Stop -> OK (wait until the 'force stop' becomes insensitive)
3. Setting -> Apps -> Motion Cam -> Memory -> Clear Cache
4. Setting -> Apps -> Motion Cam -> Memory -> Clear Data -> OK
5. Leave Safe Mode (either clock on the related notification or restart, the phone will restart in normal mode then)
6. Start Motion Cam app
7. Give related permissions
8. Start Motion Cam app

Happy to use the Motion Cam app again!

closed 2022-02-03 by giteos

Gogo1973, 2022-02-03:
> Заработих отново след известно натоварване. Следното ми свърши работа:
> 
> 1. Изключете в безопасен режим (задръжте бутона за изключване за по-дълго, след като се предлага безопасен режим, приложението Motion Cam не може да се използва в този режим)
> 2. Настройки -> Приложения -> Motion Cam -> Принудително спиране -> OK (изчакайте, докато „принудителното спиране“ стане нечувствително)
> 3. Настройки -> Приложения -> Motion Cam -> Памет -> Изчистване на кеша
> 4. Настройки -> Приложения -> Motion Cam -> Памет -> Изчистване на данни -> OK
> 5. Оставете безопасен режим (или часовник на свързаното известие, или рестартирайте, тогава телефонът ще се рестартира в нормален режим)
> 6. Стартирайте приложението Motion Cam
> 7. Дайте съответните разрешения
> 8. Стартирайте приложението Motion Cam
> 
> Радвам се да използвам приложението Motion Cam отново!

G780G

Gogo1973, 2022-02-03:
> > Имам същия телефон с A12 и няма проблем, дори винаги инсталирам приложението от тук!
> 
> Моят S20 FE е SM-G780G ... коя версия на приложението използвате?

G780G