mirsadm/motioncam #181: The choice of raw in smartphones Аsus

Jeneeeeek, 2022-02-04:
Hello, question, many Asus devices have two raw formats at once, please tell me, could you add a switch in the settings. They just have the full resolution of raw, whether it's 48/64mp , as well as the second Raw with 12/16mp binning.
[image]
in the screenshot, I showed that there are two Raw permissions at once

Jeneeeeek, 2022-02-04:
Landroid/hardware/camera2/params/StreamConfigurationMap;->getOutputSizes(I)[Landroid/util/Size;

mirsadm, 2022-02-04:
I pushed out a new beta that should pick the binned option. At some point in the future I'll add it as an option instead. Let me know if it works.

teemodk, 2022-02-04:
The beta works. Previously I had a mod to disable high resolution, but that's not needed now. Binning output is chosen for both images and "video".

mirsadm, 2022-02-05:
Cheers I'll close this for not

closed 2022-02-05 by mirsadm

Jeneeeeek, 2022-02-08:
> Ура, я закрою это, чтобы не

Zenfone 7 pro all the same has 64MP raw but doesn't have 16mp bin