mirsadm/motioncam #186: 8.0.8 - unable to change FPS

nerdpaolino, 2022-02-05:
In the last version I'm unable to change fps to 24. The file saved are all fixed to 30fps, that is a supported framerate for my device, but I was able to save in 24fps selecting from the unsupported frame rate.

HACKERALERT, 2022-02-05:
I have 15fps and 30fps as officially supported, and 25fps as "may work". But I would also like to see a 24fps option next to the 25fps option because in DaVinci Resolve, I can only export at 23.976, 24, and 30fps. 24fps is also a standard, so it would be nice to have along with 25fps and 30fps. Also, 24fps would sync better with audio at 48kHz than 25fps, which outputs a video that is a bit longer than the audio file when I drop both the DNGs and audio into DaVinci Resolve.

nerdpaolino, 2022-02-06:
I also have 30 and 15 as officially supported, but in the previous versions I was able to select and record 24 fps. NowI select 24 and the resulting file is 30.

mirsadm, 2022-03-17:
@HACKERALERT  @nerdpaolino is this still a problem with the new beta? There was a bug before where the frame rate selection would not stick if manual/shutter was selected. It should be fixed now.

nerdpaolino, 2022-03-18:
With last beta the problem is solved. Thank you very much. I close ticket

closed 2022-03-18 by nerdpaolino