mirsadm/motioncam #191: Some settings are gone after version update

LuckyBastard73, 2022-02-11:
Some of the setting (e.g. the save as DNG property) seems to be gone after updating to a new version of MotionCam via the Google Play Store.

This results in missing RAW/DNG still images in photo sessions and thus the opportunity to get more out of the images (not that the JPEGs are bad, but e.g. a white balance correction there is not such a good idea).

mirsadm, 2022-02-11:
Hmm that's interesting, it should still be there. Maybe the UI is not displaying it. Could you provide a screenshot?

LuckyBastard73, 2022-02-12:
Sorry, if I was not clear enough:
The UI elements are there but the settings after the update are not the same as before.

LuckyBastard73, 2022-02-12:
So you have a version of MotionCam installed and you made some changes in the settings
E.g. save DNG on (what is off on default)
Then you update the version of MotionCam via Google Play Store
And the settings (or at least some of them) are gone and default values again