mirsadm/motioncam #195: Problem with MotionCam Tool-0.13

Gogo1973, 2022-02-12:
Problem with MotionCam Tool-0.13, noise reduction creates a mirror strip at the edges. Watch this video: https://www.youtube.com/watch?v=Xtmu0eX4QGc

mirsadm, 2022-02-12:
Yep it extends the edges during NR but should crop them out when exporting which it is not. I need to do that.

nubbie61, 2022-02-17:
Hi. I'm having the same issue. Checked it with win-tools-12 - the same result. Looks like the dng file has been saved with this distortion BEFORE the NR from the convertor. Surely the dev would be able to finx it. Otherwise absolutely amazing product! Fantastic good job

mirsadm, 2022-02-17:
yep that's been fixed, going to push out an update soon

nubbie61, 2022-02-17:
Great news, really. Thank you very much

mirsadm, 2022-03-17:
@Gogo1973 this has finally been fixed

closed 2022-03-17 by mirsadm