mirsadm/motioncam #199: Halide error during ./setupenv.sh OSX

btilmon, 2022-02-14:
I am using macOS Mojave 10.14.6 with Clang 12.0.8. I get the following error during ./setupenv.sh for the Halide install.

-- Check for working C compiler: /Users/bjt/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/aarch64-linux-android24-clang
-- Check for working C compiler: /Users/bjt/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/aarch64-linux-android24-clang - broken
CMake Error at /usr/local/Cellar/cmake/3.22.2/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/Users/bjt/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/aarch64-linux-android24-clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/bjt/Documents/projects/motioncam/tools/convert/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make -f Makefile cmTC_a96a0/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_a96a0.dir/build.make CMakeFiles/cmTC_a96a0.dir/build
    Building C object CMakeFiles/cmTC_a96a0.dir/testCCompiler.c.o
    /Users/bjt/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/aarch64-linux-android24-clang   -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -MD -MT CMakeFiles/cmTC_a96a0.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_a96a0.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_a96a0.dir/testCCompiler.c.o -c /Users/bjt/Documents/projects/motioncam/tools/convert/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_a96a0
    /usr/local/Cellar/cmake/3.22.2/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a96a0.dir/link.txt --verbose=1
    /Users/bjt/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/aarch64-linux-android24-clang  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_a96a0.dir/testCCompiler.c.o -o cmTC_a96a0 
    ld: error: unknown argument '-search_paths_first'
    clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_a96a0] Error 1
    make: *** [cmTC_a96a0/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/bjt/Documents/projects/motioncam/tools/convert/build/CMakeFiles/CMakeOutput.log".
See also "/Users/bjt/Documents/projects/motioncam/tools/convert/build/CMakeFiles/CMakeError.log".

ishansd94, 2022-02-15:
I got the same error as well. M1 MacBook. macOS Monetary.

btilmon, 2022-02-15:
The error was specific to the NDK version! The error goes away after changing NDK version to 21.0.6113669. It got to the very end of compilation but now I am getting two missing files errors seen below. I think this is separate from the NDK issue so I will close this issue.

/Users/bjt/Documents/projects/motioncam/libMotionCam/libMotionCam/source/ImageProcessor.cpp:55:10: fatal error: 
      'exiv2/exiv2.hpp' file not found
#include <exiv2/exiv2.hpp>

/Users/bjt/Documents/projects/motioncam/libMotionCam/libMotionCam/source/RawBufferStreamer.cpp:9:10: fatal error: 
      'zstd.h' file not found
#include <zstd.h>

closed 2022-02-15 by btilmon

mirsadm, 2022-02-15:
Sorry I haven't been looking at this, just been busy with other stuff. Will try to get to it soon