mirsadm/motioncam #207: Motioncam Tools v0.11 crashes when importing the container

enprios, 2022-02-22:
Hi, first off. **Love** the motioncam app on Android. Sweet untouched pixels is beautiful.  👍

Now for the bug I encounter. 

When importing the container to my intel macbook pro (mid 2014). It crashes when I select the container in the motioncam tools app and the preview frame should render.
[image]

I couldn't find any post of this issue so I hope I don't waste anyones time if this has already been brought up.

I'll also put up this crash report
[crash.txt](https://github.com/mirsadm/motioncam/files/8116750/crash.txt)

Thanks

mirsadm, 2022-03-03:
Hey! I am aware that some people are having this issue. Thanks for providing the crash report, that's really helpful. Are you able to convert the file even if the preview does not work?

enprios, 2022-03-03:
> Hey! I am aware that some people are having this issue. Thanks for providing the crash report, that's really helpful. Are you able to convert the file even if the preview does not work?

Hi,  I can open the app, add a container but as soon I click on the list with container name it crashes. 

I hope it helps.

mirsadm, 2022-03-04:
Does this happen to you with all files? It would be good if you could record a small sample for me to test with.

enprios, 2022-03-04:
It happen on every container I try. However it worked before I updated the montioncam app I think.

 I made a super small one so I can upload. 
[VIDEO_20220304_213955.0.container.zip](https://github.com/mirsadm/motioncam/files/8188664/VIDEO_20220304_213955.0.container.zip)

mirsadm, 2022-03-07:
Thanks for providing that. I'll see if I can reproduce it.

mirsadm, 2022-03-17:
@enprios I can reproduce this, will fix for next beta release