mirsadm/motioncam #212: Remote streaming

robostep9829, 2022-03-02:
For devices with low emmc speed streaming  can be an answer. Even without wi-fi my lte reaches about 60 mbps which is enough i think(I recorded FullHD 30 fps on a phone with nice memory ~50 mbps). With ftp or custom rmtp you not only expanding bandwidth of your memory but also instantly recording file straight to your editing pc.

robostep9829, 2022-03-12:
[image]
I can even access cloud storages from a few apps, but in Motion Cam i can use only local (and Termux).

mirsadm, 2022-03-17:
@robostep9829 I'll take a look, not sure why you wouldn't be able to access it from there

robostep9829, 2022-03-27:
@mirsadm Have you heard what NDI is? I know that it's can be used in OBS for streaming over local network. I don't know if it's better than rmtp or more demanding, but it really squeezes out bandwidth for video. Also I don't know how do m.c. containers store data so maybe streaming should be implemented other way?