mirsadm/motioncam #219: S21+ 8.1.0-b0

nubbie61, 2022-03-16:
Just updated to 8.1.0-b0. MotionCam 0.15 Tools PC can't make preview nor to extract dngs - directly blows up. Previous version 013 - the same

Will have to go back to 8.0.8-b3, sadly

mirsadm, 2022-03-16:
Yes you will need the new tools to open files recorded with the new beta

nubbie61, 2022-03-16:
You mean 0.15, right? Because that's the one I used to no success

mirsadm, 2022-03-16:
0.20, they're not linked on the front page but are part of the release here:
https://github.com/mirsadm/motioncam/releases/tag/8.1.0-beta0

nubbie61, 2022-03-16:
Exactly the same I d/l and used. Sadly the exe is 0.15 inside the zip. Maybe you could check it out?

mirsadm, 2022-03-16:
https://github.com/mirsadm/motioncam/releases/download/8.1.0-beta0/motioncam-tools-windows-x86_64.zip

Does it says MotionCam Tools 0.20 in the title bar? I realise I've forgotten to update the actual version of the executable but as long as it says 0.20 in the title bar it should be the right one.

nubbie61, 2022-03-16:
Funny thing - the exe file's been compiled today - around noon time. So it should be 020, but it's not. It's 015 instead

mirsadm, 2022-03-16:
not sure what is going on, I have downloaded and converted a file with it. What version of Windows do you have? Also what CPU do you have?

mirsadm, 2022-03-16:
(also if you could send me a small container file that crashes I can try it out)

nubbie61, 2022-03-16:
Ryzen 5 5600; win 10 LTSC; all worked fine so far. I even moved back to 013/808-b3. Saved one short container. Currently upping it for you to d/l it

nubbie61, 2022-03-16:
Here's the link to the container:

https://www.jottacloud.com/s/20865e61fac518e4532b7cdf18051829188

mirsadm, 2022-03-16:
thanks I can reproduce it locally, will take a look!

mirsadm, 2022-03-16:
thanks for reporting that, found the issue and it's fixed. I'll update everything tomorrow

nubbie61, 2022-03-17:
Glad to help. Fantastic work you're doing - great thank you!

nubbie61, 2022-03-17:
Sorry for being a PITA - one more issue with the 8.1.0-b0 on S21+5G (SM-G996). In RAW video mode the start/stop button rotates non-stop; unless reset/cleared cache/data. Also - the double exposure screen is missing.

nashitahmedbarq, 2022-03-17:
> Sorry for being a PITA - one more issue with the 8.1.0-b0 on S21+5G (SM-G996). In RAW video mode the start/stop button rotates non-stop; unless reset/cleared cache/data. Also - the double exposure screen is missing.

+1. Same issue on my device, OnePlus 9R, Android 12.

mirsadm, 2022-03-17:
Do you mean the progress bar around the capture button? If you leave it for 30 seconds, does it stop?

mirsadm, 2022-03-17:
Oh nevermind I see the issue if you allocate more RAM that the default.

nashitahmedbarq, 2022-03-17:
> Do you mean the progress bar around the capture button? If you leave it for 30 seconds, does it stop?

Yes, it stops after a few seconds. (Another thing happens is that after capture, the dual exposure controls reset to their initial position)

By the way thank you for your hard work. I've never seen an android camera app with such potential.

nubbie61, 2022-03-17:
Correct - using the default RAM settings the capture button works OK.

And +1 for the app potential, highly appreciated your work

nubbie61, 2022-03-17:
Onmy setup I can't convert to dng from the app itself - it asks for the folder; then makes a new folder in there and stops doing anything; then after a while returns to the cam mode. I hope that would help you debug it

mirsadm, 2022-03-17:
@nubbie61 that'll be the same issue you're having with the tools. should be fixed once I update the app and tools, along with the infinite progress above. aiming to do it today some time

mirsadm, 2022-03-17:
@nashitahmedbarq someone requested to reset the dual exposure once a picture was taken, i think I should probably make that an option. thanks for the kind words.

nubbie61, 2022-03-17:
Appreciate all your efforts

mirsadm, 2022-03-17:
@nubbie61 could you try the latest beta? that should fix those issues

nubbie61, 2022-03-17:
Yes, thanlk you, just did and edited my comment. The all sides distortions come from the pc tools with 8.1.0-b1

nashitahmedbarq, 2022-03-18:
> @nashitahmedbarq someone requested to reset the dual exposure once a picture was taken, i think I should probably make that an option. thanks for the kind words.

That would be great. Thank you.

nubbie61, 2022-03-18:
Just to confirm the pc tools 020 produces the edge distortion on all the four edges of the frame, the in-app convert-to-dng is 100% OK, but rather slow, indeed.

mirsadm, 2022-03-18:
No problems this is very helpful. Are you able to upload a small container for me to test with?

nubbie61, 2022-03-18:
Pixel binning or edge distortion?

mirsadm, 2022-03-18:
Pixel pinning

nubbie61, 2022-03-18:
Here you are:

 

https://www.jottacloud.com/s/2081ca0ebdf38a141cba0d01fa5ad722f8e

 

5 sec, 1080p, PB, test-chart

mirsadm, 2022-03-18:
Yep that's broken! Thanks

mirsadm, 2022-03-21:
@nubbie61 this should be fixed finally

nubbie61, 2022-03-21:
Here's to confirm:
- edge hamdling has been fixed;
- pixel binning is now OK (although looks like out of focus generally)
Big thanks to you

mirsadm, 2022-03-22:
Thanks for testing that for me. I will close this for now.

closed 2022-03-22 by mirsadm