mirsadm/motioncam #221: Mi 11 Ultra with 8.1.0 beta1

FilipZamec, 2022-03-17:
Thank you for this long awaited version.

You wrote that some performance upgrades were done in this version, but I can't keep steady 60 fps anymore.

Tried i game mode, airplane mode like always, but this time no matter what I try, I usually get 8 - 20 seconds of recording. That's it.

In previous versions I got minutes, no problem.

Thank you.

mirsadm, 2022-03-17:
@FilipZamec do you remember if you set the number of compression threads to higher than 2 previously?

FilipZamec, 2022-03-17:
I kept it at default value. I don't know what it was 🤔

mirsadm, 2022-03-17:
hmm ok, there are still some more improvements to be made. The new version has some major changes to the internals but it's not quite complete yet. I'll update again soon which will hopefully restore that

FilipZamec, 2022-03-24:
Beta 2 seems to fix this issue. I can shoot 60 fps again for a long time.
Thank you