mirsadm/motioncam #231: color waves in s22 ultra

ferchubcn, 2022-04-01:
hello. when exprtar to DNG  color waves are seen. Installed the app from here and from play store. I have a galaxy s22 ultra of 256Gs

mirsadm, 2022-04-01:
hey, I think there is an issue with vignette correction on that particular phone. If you disable it, does it go away?

ferchubcn, 2022-04-01:
> hey, I think there is an issue with vignette correction on that particular phone. If you disable it, does it go away?

hey thank you very much. That was the problem. solved 👏👍

ferchubcn, 2022-04-08:
So it hasn't been resolved. Waves of colors continue to appear, with the vignette correction off

SudoFormatAll, 2022-04-10:
@ferchubcn 
the first, you must write Snap or Exynos.
And then post screenshot example of what you mean.

nurps, 2022-04-10:
Same here, rainbow colored bars in the photo, when shot into a light source (Excynos).

Where is that vignette settings, is there an advanced settings menu I need to activate/find (sorry, new to this app)

nurps, 2022-04-15:
> hey, I think there is an issue with vignette correction on that particular phone. If you disable it, does it go away?

Need anything to fix this? A dng file?

mirsadm, 2022-04-15:
@nurps yep could you send a DNG generated by the app?

nurps, 2022-04-15:
@mirsadm Here it is: https://we.tl/t-uRttcWH6ot The included preview looks fine, but the saved jpg does not.

MikkelNikkel, 2022-04-18:
Same probleme with s22 ultra exynos. Preview looks fine, but not the saved dng files, which have the same rainbow colored bars.

mirsadm, 2022-04-19:
@nurps thanks for providing that

I am looking into adding device specific profiles to work around these issues. For whatever reason the device is providing some invalid information which is why it looks like that

nurps, 2022-04-19:
It's combining 9 pixels into one (on the main cam) under the same color of the bayer matrix. 108MP Sensor, 12MP bayer matrix, maybe it has to do with that?

nurps, 2022-06-29:
> it might be providing the wrong shading map (without binning) or something weird like that.

I completely missed you made an update. In photo mode the DNG is clear now, but in burst mode it still shows that pattern, can you get that working too?

nurps, 2022-06-29:
@mirsadm Yes, just the preview, great. Thanks you.