mirsadm/motioncam #233: Instant freezing on Samsung S21

utahp123, 2022-04-07:
I can see a preview on the screen for a few seconds, then the image freezes permanently. Anybody else experiencing this, or got the app working on a base s21?

smaajlik, 2022-04-07:
> I can see a preview on the screen for a few seconds, then the image freezes permanently. Anybody else experiencing this, or got the app working on a base s21? 

Android 11? Exynos? For me working fine on S21 base exynos android 12. No bugs, no crashes.

utahp123, 2022-04-07:
Android 12, One UI 4.1, Snapdragon version. Deleted and redownloaded the
app multiple times, restarted my phone, not exactly sure what could be
causing it... Glad you've got it working on an s21 though!

smaajlik, 2022-04-07:
> Android 12, One UI 4.1, Snapdragon version. Deleted and redownloaded the
> app multiple times, restarted my phone, not exactly sure what could be
> causing it... Glad you've got it working on an s21 though!
> 
> 
> > I can see a preview on the screen for a few seconds, then the image
> > freezes permanently. Anybody else experiencing this, or got the app working
> > on a base s21?
> >
> > Android 11? Exynos? For me working fine on S21 base exynos android 12. No
> > bugs, no crashes.

Maybe try download older versions :/

danny-ray, 2022-04-09:
Hi,
the Galaxy S21 and the Galaxy z_Flip 3 5G have exactly the same problem.
The preview freezes after 3 seconds. If you switch the cam or go to the menu, it will reset for another 3 seconds.

That is an old problem, mentioned month ago and still not changed/fixed.
And yes, I took all OS/UI updates for the phone and tested all versions of motioncam. It still doesn't work.
Other apps like Filmic Pro are working great.

Seems to be an app problem.

greetz

utahp123, 2022-04-09:
Tried switching lenses and entering and exiting the menus. Got the same
result as you, unfreezes for a few seconds and then refreezes. I wonder
what is going on...

mirsadm, 2022-04-09:
This seems to be a common problem on those Samsung devices. It looks to be a firmware bug. Other people have reported the same issue with GCam for example. I'm not sure what I can do, I think Samsung has to fix it.

Pockettlens, 2022-04-20:
Hello to everyone. 
I'm using the Galaxy S21 Ultra, and I'm noticing that the app freezes, even when I lower the setting.

eduardoliron, 2022-09-12:
Same issue on Galaxy A52S 5g - Snapdragon version. Is there any updates on the issue?