mirsadm/motioncam #234: Bugs and improvements.

SudoFormatAll, 2022-04-09:
**Hi @mirsadm ! Thank you for such a wonderful application.
I thought it would never happen because of the limitations of Android, or would happen to at least after 3-5 years.
 But it works already, today, even on my old phone! I still can not believe.. I am very happy. Wish you a long life, bro.
And I also come to send you a report about mini-bugs and improvements.
So I hope you will make the app even better!**

**I have Galaxy S10E Exynos:**
1. Premiere 2020/2022 does not open DNG files for some reason, writes that it is damaged. Despite the fact that even Photoshop 2019 opens them without problems. can it be fixed? Raw converter from Adobe wider than in Davinci. also you can use rich ACR settings, including Raw-profiles.
2. Exposure have some bug. For example, I set 1/48 and in DNG file it is showed as 1/47. I can't check it on the eye, is simply an info error or a real settings bug.
3. The aperture is displayed as 1.7 for 1.5 and 2.4 the same. (BUG)
4. change ISO with jumps like pro cameras are not very good. add the option to be able to change ISO regulator with scroller to choose any value.
5. Exposure values ​​must be not locked with all "default" values, but must be based on shutter speed and duplicate this value. (for example, 24fps - 1/24 1/48 etc.) Now too much extra values, spending times.
6. FPS works not correctly (BUG). For example, on the selected 24fps - shoot 19-20.
On the chosen 48fps - shoot 23-24. and for 120FPS - stable 24fps. 
  6.1 But when choosing 120fps (unsupported) - the focus is stopped working, for all. auto and manual and the tap.
  I have a choice - to rec a stable 24fps, but without focus, or rec only 22fps, but with a focus... a bad choice.
  (does not depend on the size of the video resolution, and from ram. Ram is not crowded)
  6.2 also when choosing 60fps (unsupported) - the program freezes.
7. manual focusing lacks focus picking.
8. needs to remove the brightness strip from the screen when the auto mode is not selected. It is vastly space and only interferes.
9. The selected aperture is not highlighted. It is necessary to make it highlighted in blue, like the other parameters when you choose them manually.
10. camera cannot create a folder for video if it was deleted, and save files to root-data. It is very stupid and uncomfortable. Give it rights to create a folder selected to save the RAW-video if it is missing.
11. The "Auto" button is needed for each parameter for reset it. For example, if I use a manual focus, and I want to return to the auto without stopping the video, I can't do it. now i have only option to reset to auto everything. This is bad.
12. Now only two (4k and 1080p) cropping presets, need to add the funx to save "my crop preset".
13. Strips displayed Ram and Storage on the bright Sun are not visible, their translucent border, so it is not make sense where the real end of line.
 Adding % to the end of strip on the side will solve this problem.

**Converter (Motioncam-Tools-Windows):**
1. You need to change the order in rendering to "remove vignette" and after "stack".
The magenta vignette appears in the dark, which is not visible to the eye generally, in stacking even from 4 photos, it becomes so much pink that closes half the pic.
first need to remove the pink vignette and then do stack, without this "pink" information.
2. Lacks presets. It is very uncomfortable to set up the sliders, and if there are many files, it is very big pain. one by one....Need to do presets and apply to all.
3. Lacks for "delete only one file" from the queue.
4. need to add the minimum value 2 photos for stack. bright sun shots or with "Binnig" mode is selected need this. Two frames are enough to not lubric microdetal in such scenarios.

Let me know if I can still help with another improvement of the app!

mirsadm, 2022-04-20:
@SudoFormatAll Hey,

Sorry for taking so long to respond. Thanks for the feedback will look into these

SudoFormatAll, 2022-04-21:
@mirsadm 

ok! cant wait for update! :)

SudoFormatAll, 2022-04-26:
@mirsadm 
i upd report with some correction and new two bugs. please not loose them.

SudoFormatAll, 2022-08-14:
@mirsadm hi, how is it going? 
a lot of time has passed. is there any progress?

p/S i heard that you said in some announcement that you "closed" here project because you get no reviews and ideas, but I came recently and immediately gave you a bunch of ideas ... but so far not one of them has been completed and the project is "closed" ..