mirsadm/motioncam #237: windows converter is deleting wav files after container conversion

2blackbar, 2022-04-13:
so there is no way to get the sound now, i have to re-copy them from backup, imo its a big deal.
i copied container files and wav files to ssd, converted container files to dng on windows, picked no compression and no vignette correction, after conversion my wav files were gone from ssd.

mirsadm, 2022-04-20:
hey, the tools don't ever delete files, I'm not sure what went wrong there

2blackbar, 2022-04-24:
Just to be safe i move all wav files to separate folder after it happened for 3rd time, it doesnt always happen so theres particular recipe for it to happen, i did not figure it out yet

SudoFormatAll, 2022-05-04:
@2blackbar im on windows10, and tools never and nothing delete...files always stay in place you put them...
tools dont even choice wav files,,, im sure you do a crazy things.

or just do the video what you do to proof whats happening...
im in 99% sure u can find your mistake yourself when you see your video yourself..