mirsadm/motioncam #239: Broken red channel

HellbringerOnline, 2022-04-17:
Hello. I am trying to use this app on my galaxy tab a7 and i am getting broken red channel in dng files. But I have normal channels on Jpeg image. How can I fix this?

jpeg
[image]
dng
[image]
Broken Red
[image]

mirsadm, 2022-04-17:
Hi,
Is that when you take a photo in the app?

HellbringerOnline, 2022-04-17:
> Is that when you take a photo in the app?
Yes, this is a preview of jpeg from the app and dgn from the app
I have another photo where this effect is less, but it is still there
[image]

mirsadm, 2022-04-20:
hmm that is pretty weird, the red channel seems to be clipped. are you able to record a very small RAW video for me to test with?

HellbringerOnline, 2022-04-20:
> hmm that is pretty weird, the red channel seems to be clipped. are you able to record a very small RAW video for me to test with?

Did you mean DGN frames?

2blackbar, 2022-04-24:
no, just container files, 1 second or something, and upload to googledrive then post link here

HellbringerOnline, 2022-04-24:
> no, just container files, 1 second or something, and upload to googledrive then post link here

https://drive.google.com/file/d/1-fNe1NnbqdyqWnYKEiCWgK7__HsEHgzp/view?usp=sharing