mirsadm/motioncam #240: Can we have separate fix for vignetting(dark corners) and separate for pink tint in the middle?

2blackbar, 2022-04-20:
I noticed that by fixing vignette enabled you still lose some dynamic range in highlights , but all i want to do is to nullify pink color in the middle , its more visible in dark scenarios, its probably part of vignette.
It would be great if we could preserve dynamic range and get rid of pink tint only, without affecting highlights.

2blackbar, 2022-04-20:
I think its a vignette related cause the pink circle is slightly stronger when less light hits the lens,, Its not jarring, its very subtle pink circular shaped tint , ive seen it on other phones its much stronger than on mine, maybe a slider with vignette fix strength would be universal solution for all phones out there so people could just set it up for their phones, but mauybe its not how the fix works i dont know.
I was just wondering if this can be split from the whole vignette fix, so id just fix the tint and leave exposure , highlights/shadow values as it was.
You can see the tint on this one https://www.youtube.com/watch?v=2a7Mv0NE_OI
Look at the white part of the clouds, theyre pinkish white , but once that cloud is on the side of the frame, it becomes more neutral white.That slight tint happens to everything thats in the middle of the frame but its most visible on white and on black.

mirsadm, 2022-04-21:
@2blackbar I think I may be able to solve this problem. I have been able to replicate it on my phone by overexposing to the point the camera is completely saturated. Will keep you updated.

2blackbar, 2022-04-22:
Great, i can provide container file if needed

mirsadm, 2022-04-22:
yes that would be helpful actually, thanks!

2blackbar, 2022-04-22:
here it is, you can see pink tint in the middle , when paninng and paper moves to the side, then that part of the paper that was pink, is no longer pink, its also affecting black a bit
https://drive.google.com/file/d/1UArmYa0KlY_3U6Aufm5yLQqJx5jWQOYS/view?usp=sharing

mirsadm, 2022-04-24:
thanks will use it for testing