mirsadm/motioncam #242: Fake-Details (Noise Reduction area miscalculation)

SudoFormatAll, 2022-04-26:
I don't know how to describe this bug correctly. it is difficult to see it. 100% zoom.
if the photo has a lot of noise (low light / night) - the "stack" does not remove all noise, and in some places there are poorly visible, but large squares in which the noise is not removed at all. 
this would not be a problem if later when we want to kill the noise with the noise reduction software (I use nit video) - it sees these squares as details, and not "clean" it. so in the end we get only half of the picture with the noise removed, it looks terrible. 
and than higher value of "micro-details" in RAW settings, then more distinct these squares.
can this be fixed somehow? 

I attach a crop of a night frame with an example of such squares...
one - before noise reduction, the second with noise reduction

P.S. also, because of the compression in jpeg, it is a little smeared, in the original it a little bit clear contours.

P.S.S. no spoiler, so i put links to not clog layout.

https://user-images.githubusercontent.com/103313570/165347881-53229a39-aaf0-4ffd-9f62-99454969574e.jpg
https://user-images.githubusercontent.com/103313570/165347869-fa163bd6-4996-40e0-914f-bae20bda63ab.jpg
https://user-images.githubusercontent.com/103313570/165347889-832977dc-7ab8-496f-87aa-a5c98041a390.jpg
https://user-images.githubusercontent.com/103313570/165347901-743bf579-dde9-420c-94aa-5b926aeab843.jpg
https://user-images.githubusercontent.com/103313570/165347910-74935e5c-0264-4259-ada2-914b8c40203e.jpg
https://user-images.githubusercontent.com/103313570/165347916-cc0683e8-5aa2-41a8-a0c4-442f814bc3fc.jpg

mirsadm, 2022-04-26:
How many frames do you stack?

SudoFormatAll, 2022-04-26:
@mirsadm the possible minimum is always 4.
 as I already wrote in another topic, 2 are needed for pictures in the sun, and 4 is the maximum working value.
 anything more is too much killing of micro-details and making everything blurry. more-more better use neat-video after stack.

mirsadm, 2022-04-26:
When there's that much noise there isn't much you can do. You need to increase the number of frames to stack. The reason is that it merges them in blocks. Naturally some will overlap which means parts of the image will have less noise reduction applied. If you increase the number of images the problem gets mitigated because there's more data to work with. In the future I'll provide more ways to tweak how the stacking works so you might be able to get better results depending on the scene.

SudoFormatAll, 2022-04-26:
@mirsadm Firstly, please read more carefully, I wrote that the problem does not depend on the noise. This problem is during the day, in bright photos, it is simply more difficult to notice it. but it there on some places,,
Secondly, even if you increase the stack, this is not a way to solution. The problem of improper determination of the area of ​​parts. area miscalculation. Squares must be smaller. that is, more accuracy and precision.

Later I will try to make Stack with this raws in Photoshop. I have never seen such problems in PS.
OOOR, either there is an assumption that this is a demosaic bug, and that the initial img already have such squares as lost/broken information, it is simply not visible until the information from these areas is pulled out.
I will do more some test of a slightly deeper with Photoshop, in next few days.

mirsadm, 2022-04-27:
@SudoFormatAll as I said above the problem is how just how the stacking works. There lots of variables to tweak behind the scenes that are being automated for simplicity. I think in your case if you want to achieve the best result you would need to tweak them manually. At some point I will add that as an option.

SudoFormatAll, 2022-04-27:
@mirsadm maybe you are right and additional settings will solve the problem... maybe
I just thought that the stack is a banal and the same operation in any software. +/- same.
yes there are many modes, but for noise reduction, they are not that much different. i use all modes and never get this problem, in PS, and even in android when i use stack in gCam with 2 or more shots... so this is a reason why i very surprise that we need some additional settings to solve it....

also, I did an experiment and made a stack with the same photos in photoshop
and there is no such problem with square artifacts. even more, there is no pink in the shadows.
I didn't use any other settings, I just took the first 4 frames and stacked in "mean" mode
CamTools: https://user-images.githubusercontent.com/103313570/165526013-2a7908cd-7b9c-4748-acd3-b637cb069dec.jpg
PS: https://user-images.githubusercontent.com/103313570/165526052-3ad4fa7b-213a-4dfa-84b7-ecf480a39db1.jpg

and one crop with PS stack + Neat-video applied, for comparison with the shot from the first post. 
https://user-images.githubusercontent.com/103313570/165526233-39b42761-77b6-4d5c-8973-23d1809990a8.jpg

SudoFormatAll, 2022-04-30:
@mirsadm I will leave here one more shot, but in day bright light.
just to prove that the problem exists not only at night, and does not depend on the amount of noise.

offc minimum 4 shots stack.
https://user-images.githubusercontent.com/103313570/166097609-df1f378e-340d-45e3-bb8a-fe77362a3a26.jpg

SudoFormatAll, 2022-08-14:
@mirsadm hi, how is it going? 
a lot of time has passed. is there any progress?

p/S i heard that you said in some announcement that you "closed" here project because you get no reviews and ideas, but I came recently and immediately gave you a bunch of ideas ... but so far not one of them has been completed and the project is "closed" ..