mirsadm/motioncam #251: Wrong "White Balance" on OnePlus 9 Pro - MotionCam Pro

Muckymuckman, 2022-05-30:
On any other version of MotionCam after version 8.0.8 - beta1, there is a problem with white balance. When viewing the DNG with any Camera Raw (on a PC or in Lightroom), you can observe that the Tint slider is shifted to the left edge, with a tinge of purple always present. In dng from MotionCam Pro it is literally impossible to get the white balance right. 
The problem is not repeated in MotionCam 8.0.8 - beta1.
I have attached a couple of screenshots of dng files taken with different versions of MotionCam + 2 dng frames from different versions in the archive.

[image]
[image]
[dng frames.zip](https://github.com/mirsadm/motioncam/files/8799040/dng.frames.zip)

Muckymuckman, 2022-05-30:
dng frames.zip

mirsadm, 2022-05-30:
@Muckymuckman thanks for reporting that. Are you able to record a short RAW video and share it? That'll be helpful in finding the issue

Muckymuckman, 2022-05-30:
@mirsadm 
Done. Attached are two links to archives with recordings
https://drive.google.com/file/d/13D1y2AdtK5VDCOIIKZlVpsgip5N-cLip/view?usp=sharing
https://drive.google.com/file/d/1XRcRRe9A4KtxHYATnslN2x4sdLGRVzPV/view?usp=sharing

mirsadm, 2022-05-31:
@Muckymuckman thanks for providing those. I have found the reason why the white balance isn't right. Weirdly it seems to be an issue with just LR (Google Photos and my Mac show the correct white balance). Anyway I will push out a fix to correct this hopefully today.

Muckymuckman, 2022-05-31:
@mirsadm Awesome. Looking forward to it!

mirsadm, 2022-05-31:
@Muckymuckman thanks again for reporting that issue. I've pushed an update with the fix to the play store. You can contact me directly on Discord if you encounter any issues!

closed 2022-05-31 by mirsadm