mirsadm/motioncam #254: Unusable on ZTE Axon 30 Ultra in Version 9.0.5

LuckyBastard73, 2022-06-19:
Not sure what happened but the new version 9.0.5 is not usable on the ZTE Axon 30 Ultra anymore.
[IMG_20220619_200302.zip](https://github.com/mirsadm/motioncam/files/8936107/IMG_20220619_200302.zip)
[IMG_20220619_200302.jpeg.zip](https://github.com/mirsadm/motioncam/files/8936108/IMG_20220619_200302.jpeg.zip)

I was so happy what MotionCam could do and now that...

mirsadm, 2022-06-19:
What is wrong with the DNG in the current version? Is it the stacking that's not as effective?

LuckyBastard73, 2022-06-19:
There is noise like hell. In the DNG and in the JPEG.
There is a setting "Apply noise reduction to DNG", is the also affecting the JPEg result? If so the label should be changed.
When I turn this on the images look better. But compared to the old ones very flat and overexposed.
[IMG_20220619_203302.jpeg.zip](https://github.com/mirsadm/motioncam/files/8936157/IMG_20220619_203302.jpeg.zip)
[IMG_20220619_203256.dng.zip](https://github.com/mirsadm/motioncam/files/8936161/IMG_20220619_203256.dng.zip)

I was really happy with the results from the old version, with the results of the new version I am completely unhappy. On our last vacation beginning of the week I was already wondering why the images from ProShot looks better than the ones fromMotionCam inlow light. So maybe the right settings have changed. I was not expecting that.

LuckyBastard73, 2022-06-19:
I am talking about the still image section...

mirsadm, 2022-06-19:
Apply noise reduction to DNG was enabled by default before. Otherwise the noise reduction would only apply to the JPEG. Are you referring to the JPEG or the DNG results that have changed? The JPEGs will look different because I've changed the tonemapping. I will probably change this back though. If it's the DNG then that should be exactly the same, minus what I was referring to before about the noise reduction

LuckyBastard73, 2022-06-19:
The JPEGs should not get noisier by applying some tone mapping.
Maybe you can have a look at the attached DNGs and JPEGs.
In my opinion the JPEGs are now way too flat and overexposed.

mirsadm, 2022-06-19:
Are you capturing these in burst mode and choosing the noise reduction settings yourself? It it's in auto mode it may just be not picking the right settings automatically

LuckyBastard73, 2022-06-19:
I used normalmode. Night and Photo.
My impression is that the difference between the two modes (turned off auto night mode) was more pronounced in the old version.

Whith the new version I really struggle to get results that are pleasing to my eyes.

mirsadm, 2022-06-19:
I think the flatness is due to the tonemapping change, it isn't quite as contrasty. Keep in mind you can change how it looks by changing the photo settings in the arrow menu. Also, if you disable the basic viewfinder you will get an accurate representation of what the image will look like as well as having a slider to control the shadows/tonemapping. 

The photo mode will improve over time but it has taken a couple of steps temporarily because most people are using the the app for the video recording. There will be an update coming soon which will improve the colour reproduction and then I'll slowly get it back to hopefully be better than it was before.

If you are able join us on Discord. You can post bugs/ideas/suggestions etc. It's quite helpful for me when it's posted there.

Discord link:
https://discord.gg/N7wWtzuj2A

mirsadm, 2022-06-19:
* I think the main issue is that the app use to adjust the shadows automatically. That is disabled at the moment while I rework it. So you won't be able to get great results without disabling the basic viewfinder and controlling the shadows manually.

LuckyBastard73, 2022-06-19:
Well maybe too many changes form me to check all the implications.
Thank you for the hint with the basic viewfinder. Made a first test. Think I know how to use it for better results.
Im am not sure how many people that are not developers check what is behind this concept of basic and non basic viewfinder.

An app with geek settings is fine, so that eyperts can adjust as much as required. The default mode on the other hand should be as dummy friendly as possible to get a good trade off right out of the box.

I am excited for the changes to come...

mirsadm, 2022-06-19:
You are right so one of the next steps for me is to add help buttons and explanation for everything. In general I want to move away from offering a completely automatic experience. The reason is that the stock camera on many phones does that already and it's impossible to please everybody with an automated mode. Instead I want to provide simple tools to be able to customise the image in a basic way. The shadows slider is one way that lets you control how the image will look. It's not quite there yet though. It should provide a good starting estimate for you to adjust from. Right now it is not doing that.

LuckyBastard73, 2022-06-30:
Hi, I have seen it and first impression is positive although I did not have the time to investigate a little deeper.
I am happy enough to be a user of the Pro version of the app as of now. Not a huge financial support but at least a small one.

I don't know if you have seen it but I was not so happy with the place of the module switching element as it is placed now as I am a right hander. Maybe a switch to swap the elements (pro features and module change) would be a great enhancement.

And speaking of improvements ProShot introduced HEIC image support some time ago and I use it quite a lot. Is there a chance to get this as well (for chipsets that supports the encoding in hardware)? I know that this app is now more focused on video than on still images but for me the poto pat is the more interesting one.